lovyan03/LovyanGFX #734: Error: 'gpio_hal_iomux_func_sel' was not declared in this scope

brightproject, 2025-07-17:
## Environment ( 実行環境 )

- MCU or Board name:  ESP32-S3
- Panel Driver IC:    Panel_ST77916
- Bus type:           QSPI
- LovyanGFX version:  1.2.9
- FrameWork version:   Arduino Core 3.3.0-RC1, ESP-IDF v5.5.0
- Build Environment:   PlatformIO
- Operating System:   Windows

## Problem Description ( 問題の内容 )

Trying out a new version of the framework:

> platform = 
>   ; Arduino Core 3.3.0-RC1 based on ESP-IDF v5.5.0 for Platformio - Pre-release
>   https://github.com/pioarduino/platform-espressif32/releases/download/55.03.30-rc1/platform-espressif32.zip

I get a compilation error, although I don’t use the RGB panel at all at the moment:

> Dependency Graph
> |-- LovyanGFX @ 1.2.9
>
> PACKAGES:
>  - contrib-piohome @ 3.4.4
>  - framework-arduinoespressif32 @ 3.3.0
>  - framework-arduinoespressif32-libs @ 5.5.0+sha.cf8dad0746
>  - tool-esptoolpy @ 5.0.1
>  - tool-mklittlefs @ 3.2.0
>  - toolchain-xtensa-esp-elf @ 14.2.0+20241119
> 
> Compiling .pio\build\PROJECT\lib16c\LovyanGFX\lgfx\v1\platforms\samd51\Bus_I2C.cpp.o
> Compiling .pio\build\PROJECT\lib16c\LovyanGFX\lgfx\v1\platforms\samd51\Bus_SPI.cpp.o
> lib/LovyanGFX/src/lgfx/v1/platforms/esp32s3/Bus_RGB.cpp: In function 'void lgfx::v1::_gpio_pin_sig(uint32_t, uint32_t)':
> lib/LovyanGFX/src/lgfx/v1/platforms/esp32s3/Bus_RGB.cpp:98:5: error: 'gpio_hal_iomux_func_sel' was not declared in this scope; did you mean 'gpio_hal_func_sel'?
> Compiling .pio\build\AVIA_HORISON\lib16c\LovyanGFX\lgfx\v1\platforms\samd51\common.cpp.o
>    98 |     gpio_hal_iomux_func_sel(GPIO_PIN_MUX_REG[pin], PIN_FUNC_GPIO);
>       |     ^~~~~~~~~~~~~~~~~~~~~~~
>       |     gpio_hal_func_sel

In the previous version everything worked fine:

> platform =
> https://github.com/pioarduino/platform-espressif32

Basitadam, 2025-07-24:
Getting the same error while using RGB panel:

- MCU or Board name: ESP32-S3
- Panel Driver IC: Panel_RGB
- Bus type: Bus_RGB
- LovyanGFX version: 1.2.7
- FrameWork version: Arduino Core 3.3.0 , ESP-IDF v5.5.0
- Build Environment: Arduino IDE 2.3.6
- Operating System: Windows

brightproject, 2025-07-25:
https://github.com/lovyan03/LovyanGFX/issues/714#issuecomment-2968349655

AcuarioCat, 2025-08-24:
I have tested this fix on the Elecrow 7" HMI display and it appears to work correctly.

brightproject, 2025-09-20:
> hi,
> 
> this commit on the develop branch fixes the compilation error, but remains untested
> 
> [ac9198d](https://github.com/lovyan03/LovyanGFX/commit/ac9198dfd8a52421cac424e1473309671b09037d)

```c++
void gpio_iomux_out(uint8_t gpio_num, int func, bool out_en_inv) __attribute__((deprecated("Please use `gpio_iomux_output` instead")));
``` 

I set in `lib\LovyanGFX\src\lgfx\v1\platforms\esp32s3\Bus_RGB.cpp`
```c++
  static void _gpio_pin_sig(uint32_t pin, uint32_t sig)
  {
    // https://github.com/lovyan03/LovyanGFX/issues/714
    // https://github.com/lovyan03/LovyanGFX/issues/734
    // gpio_hal_iomux_func_sel(GPIO_PIN_MUX_REG[pin], PIN_FUNC_GPIO);
    // gpio_iomux_out(pin, PIN_FUNC_GPIO, false);
    gpio_iomux_output(pin, PIN_FUNC_GPIO, false);
    gpio_set_direction((gpio_num_t)pin, GPIO_MODE_OUTPUT);
    esp_rom_gpio_connect_out_signal(pin, sig, false, false);
  }
```