lovyan03/LovyanGFX #740: 【バグ報告】developブランチで Panel_ST7701.hpp / Bus_RGB.hpp など一部ファイルが消失しています

UnnamedJOE, 2025-08-02:
お世話になっております。

現在（2025年8月2日）、developブランチにて
- src/lgfx/v1/panel/Panel_ST7701.hpp
- src/lgfx/v1/bus/Bus_RGB.hpp
など一部のファイルが消失しており、Web UI上で見えても404エラーとなります。
また、ZIPでダウンロードしても同様にこれらのファイルが含まれていません。

Arduino開発等で困っている方が多数いると思います。
ご確認・ご対応いただけますと幸いです。

lovyan03, 2025-08-03:
すみません、それらのファイルがいつの時点で存在したか教えて頂けますか？
ご指定のパスにはそもそも作った覚えがないです。

UnnamedJOE, 2025-08-03:
なんかご迷惑おかけしてすみません。
ChatGPTに誘導された先の一覧に該当ファイルのリンクがあったのですが仕組みそのものを理解しておらず、ChatGPTの指示で報告すれば良いとの事で報告してしまいました。
作者様が作られた覚えの無い物＝ChatGPTが作った偽リンクなんだと思います。
この度はお手数お掛けして申し訳ありませんでしたm(_ _)m

[image]
[image]

lovyan03, 2025-08-04:
AIはもっともらしく間違った情報を提示することがざらにあるので、必ずご自身で裏付けを取ることをおすすめします。

tobozo, 2025-08-04:
Hi,

found a Panel_ST7701 object hidden in the platforms/esp32s3 folder

https://github.com/lovyan03/LovyanGFX/blob/5438181440c71cf30bbdc347b0b1597ae3ebf77d/src/lgfx/v1/platforms/esp32s3/Panel_RGB.hpp#L96-L105

UnnamedJOE, 2025-08-04:
Hi @tobozo,

Thank you very much for your comment!

I had been struggling to find the `Panel_ST7701.hpp` file, so your information was incredibly helpful. I just checked the `platforms/esp32s3/Panel_RGB.hpp` file and indeed found the hidden `Panel_ST7701` reference. I really appreciate you pointing it out!

If you have any additional advice on how to properly initialize and use this hidden `Panel_ST7701` object with ESP32-S3 (especially with a 4.3-inch ST7701 display), I’d be grateful.

Thanks again for your support!