lovyan03/LovyanGFX #743: No output with SPI display on ESP32 P4, bus stall with pushPixels - same code works on S3.

PhonicUK, 2025-08-05:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name:  Waveshare ESP32 P4 Nano
- Panel Driver IC:  ST7796
- Bus type:  SPI
- LovyanGFX version:  Development
- FrameWork version:  ESP-IDF 5.4.1
- Build Environment:   VSCode
- Operating System:   Windows

## Problem Description ( 問題の内容 )

When using the ESP32-P4 to drive a ST7796 - I get no output for basic operations such as `fillScreen` - and using `pushPixels` will cause an outright stall with no data being sent and the execution stopping with a code of `0xdeadc0de`

My exact same code (minus the pinout differences) works perfectly on an S3 board.

### Expected Behavior ( 期待される動作 )

The same code works on an S3 and a P4.

### Actual Behavior ( 実際の動作 )

The P4 doesn't clear the display and is unable to push pixels.

### Steps to reproduce ( 再現のための前提条件 )

1. Configure an ESP-IDF project for the P4
2. Wire a ST7796 to free GPIO pins
3. Attempt to drive the display

I have set up my environment with the following parameters:

```
      cfg.spi_mode = 1;             // SPI通信モードを設定 (0 ~ 3)
      cfg.freq_write = 20000000;    // 送信時のSPIクロック (最大80MHz, 80MHzを整数で割った値に丸められます)
      cfg.freq_read  = 16000000;    // 受信時のSPIクロック
      cfg.spi_3wire  = false;        // 受信をMOSIピンで行う場合はtrueを設定
      cfg.use_lock   = true;        // トランザクションロックを使用する場合はtrueを設定
      cfg.dma_channel = SPI_DMA_CH_AUTO; // 使用するDMAチャンネルを設定 (0=DMA不使用 / 1=1ch / 2=ch / SPI_DMA_CH_AUTO=自動設定)
      // ※ ESP-IDFバージョンアップに伴い、DMAチャンネルはSPI_DMA_CH_AUTO(自動設定)が推奨になりました。1ch,2chの指定は非推奨になります。
      cfg.pin_sclk = 46;            // SPIのSCLKピン番号を設定
      cfg.pin_mosi = 47;            // SPIのMOSIピン番号を設定
      cfg.pin_miso = -1;            // SPIのMISOピン番号を設定 (-1 = disable)
      cfg.pin_dc   = 54;            // SPIのD/Cピン番号を設定  (-1 = disable)
...
      cfg.pin_cs           =    53;  // CSが接続されているピン番号   (-1 = disable)
      cfg.pin_rst          =    48;  // RSTが接続されているピン番号  (-1 = disable)
      cfg.pin_busy         =    -1;  // BUSYが接続されているピン番号 (-1 = disable)
...
      cfg.pin_bl = 45;              // バックライトが接続されているピン番号
```

And wired my display up appropriately.

Right now I am trying to just get the display to initialise and clear, so I am running the following:

### Code to reproduce this issue ( 再現させるためのコード )

```cpp
  tft.begin();
  tft.setBrightness(255);
  tft.fillScreen(TFT_BLUE);
```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

tobozo, 2025-08-05:
hi,

P4 + Bus_SPI have only been tested with ILI9341 so far, so your feedback is very useful 👍 

Please make sure you're using the develop branch of this repository.

There are a few things you may try:

```c
// cfg.spi_mode = 1;  // you don't need this but P4 works with 0, 2, 3
// cfg.freq_read  = 16000000; // you don't need this since miso = -1
// cfg.spi_3wire  = false; // you don't need this unless using tft.readRect(), which wouldn't work since miso = -1
// cfg.use_lock   = true; // not sure if you need this 
cfg.spi_host = SPI2_HOST; // you may need this
```

Unrelated but here's the configuration that was tested with Olitec P4 devboard and a generic ILI9341+Touch board:

```cpp
class P4Display_ILI9341 : public lgfx::LGFX_Device
{
  lgfx::Panel_ILI9341 _panel_instance;
  lgfx::Bus_SPI       _bus_instance;
  lgfx::Touch_XPT2046 _touch_instance;
  lgfx::Light_PWM     _light_instance;

  public:

    P4Display_ILI9341()
    {

      {
        auto cfg = _bus_instance.config();
        cfg.freq_write = 80000000;
        cfg.freq_read  = 80000000;
        cfg.pin_mosi = GPIO_NUM_53;
        cfg.pin_miso = GPIO_NUM_54;
        cfg.pin_sclk = GPIO_NUM_5;
        cfg.spi_host = SPI2_HOST;
        cfg.pin_dc   = GPIO_NUM_21;
        //cfg.spi_mode = 0 ; // works with 0, 2, 3
        _bus_instance.config(cfg);
        _panel_instance.bus(&_bus_instance);
      }

      {
        auto cfg = _panel_instance.config();
        cfg.pin_cs   = GPIO_NUM_6;
        cfg.pin_rst  = GPIO_NUM_4;
        cfg.readable = true;
        _panel_instance.config(cfg);
      }

      {
        auto cfg = _touch_instance.config();
        auto buscfg = _bus_instance.config();
        cfg.bus_shared = true;
        cfg.pin_mosi = buscfg.pin_mosi;
        cfg.pin_miso = buscfg.pin_miso;
        cfg.pin_sclk = buscfg.pin_sclk;
        cfg.spi_host = buscfg.spi_host;
        cfg.pin_cs   = GPIO_NUM_20;
        _touch_instance.config(cfg);
        _panel_instance.setTouch(&_touch_instance);
      }

      {
        auto cfg = _light_instance.config();
        cfg.pin_bl = GPIO_NUM_2 ; // Backlight pin
        //cfg.invert = false;
        _light_instance.config(cfg);
        _panel_instance.setLight(&_light_instance);
      }

      setPanel(&_panel_instance);

    }
};

```

PhonicUK, 2025-08-05:
Ah so I've made a little progress, I've got it to show a solid color screen using `fillScreen` - I'm not entirely sure what I changed but it's clearly able to write to the display in some form now. `fillRect` also works.

Using pushPixels still causes the exact same problem. It stalls out entirely on writeBytes:

```c++
  tft.begin();
  tft.setBrightness(255);
  
  tft.fillScreen(TFT_RED);
  tft.fillRect(128,128,64,64,TFT_WHITE);

  tft.startWrite();
  uint16_t testPixels[64];
  for (int i = 0; i < 64; i++) {
    testPixels[i] = (rand() % 65536);
  }
  tft.setAddrWindow(32, 32, 64, 64);
  tft.pushPixels(testPixels, 64);
  tft.endWrite();
```

This produces a white square on a red background, but the test data sent using pushPixels causes the same stall and trace as above. Seems very weird to me that it can send over solid fills like that but not raw pixel data.

I'm using the latest Develop branch:

```powershell
PS R:\source\jet32\components\LovyanGFX> git pull
Already up to date.
PS R:\source\jet32\components\LovyanGFX> git status
On branch develop
Your branch is up to date with 'origin/develop'.
```

I've tried using `SPI2_HOST` and tried a few different pin arrangements, but get essentially the same result.

For reference, here is the pinout for the particular board I'm using:

[image]

As you can see I'm using the GPIO pins from the right side of the board: 45-48, 53, 54. I did see the config you've posted from the other thread but it doesn't help me much here other than knowing that SPI2_HOST is usable.

tobozo, 2025-08-05:
enclosing pushPixels between tft.startWrite() and tft.endWrite() sounds redundant

https://github.com/lovyan03/LovyanGFX/blob/5438181440c71cf30bbdc347b0b1597ae3ebf77d/src/lgfx/v1/LGFXBase.hpp#L342-L344

what about pushImage ?

```cpp
  uint16_t testPixels[64];
  for (int i = 0; i < 64; i++) {
    testPixels[i] = (rand() % 65536);
  }
  tft.pushImage(32, 32, 8, 8, testPixels );
```

tobozo, 2025-08-05:
same with pushImageDMA and pushPixelsDMA ?

maybe now is a good time to play with cfg.spi_mode and other bus config values

PhonicUK, 2025-08-05:
Tracked it down to here: https://github.com/lovyan03/LovyanGFX/blob/d0499d03a15e0a3b9b71e0771c0492f730dff3b4/src/lgfx/v1/platforms/esp32/Bus_SPI.cpp#L708

This loop never exits if you feed more than 32 bytes. If I comment out this loop, then I can draw sprites that are 64x64px - band setAddrWindow / pushPixels or pushImage work *sometimes*, but not their DMA equivalents. Those versions just write 0x0000.

Trying to write out a single *field* in my case (480x1) results in just the first 32 pixels being sent though.

tobozo, 2026-02-09:
@weiernt thanks for investigating this 👍 

You'll probably have to apply similar changes in `platforms/esp32/common.cpp`.

Can you please share your LGFX config, particularly the Bus_SPI configuration block, I want to compare with the test config I use ([ESP32-P4-Devkit from Waveshare](https://www.waveshare.com/wiki/ESP32-P4-Module-DEV-KIT-StartPage) + [LCD 3.5 Hat](https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(A))) 

```cpp
    {
      auto cfg = _bus_instance.config();
      cfg.spi_host = SPI2_HOST;
      cfg.spi_mode = 0;
      cfg.freq_write = 15000000;
      cfg.freq_read  = 16000000;
      cfg.spi_3wire  = true;
      cfg.use_lock   = false;
      cfg.dma_channel = SPI_DMA_CH_AUTO;
      cfg.pin_sclk = 0;
      cfg.pin_mosi = 3;
      cfg.pin_miso = 2;
      cfg.pin_dc   = 4;

      _bus_instance.config(cfg);
      _panel_instance.setBus(&_bus_instance);
    }

```

weiernt, 2026-02-11:
@tobozo There's a minor problem in the code that was merged where if the assigned DMA channel to the SPI bus isn't 0, then the reg_write to link2 would be the wrong link2 register. 

You can refer to my new commit here https://github.com/weiernt/LovyanGFX/commit/bd6f186d2104428c51e61818cd75da1f572dc942 

There's probably a more elegant way to do it, so feel free to modify it.