lovyan03/LovyanGFX #745: ILI9488 `writecommand` not working for my setup

TokyoExpress, 2025-08-12:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name:  ESP-WROOM-32
- Panel Driver IC:    ILI9488
- Bus type:           SPI
- LovyanGFX version:  latest develop
- FrameWork version:   ArduinoESP32? not sure
- Build Environment:   ArduinoIDE
- Operating System:   Windows

## Problem Description ( 問題の内容 )

`tft.writecommand()` is not working, I am trying to get TE pin to toggle on but no commands seem to work. As an example:

The display initializes but never sleeps or turns off.

### Expected Behavior ( 期待される動作 )

Sending hex commands results in desired behavior (e.g. display off, TE line on)

### Actual Behavior ( 実際の動作 )

Sending hex commands does not have any effect

### Code to reproduce this issue ( 再現させるためのコード )

**Please submit complete source code that can reproduce your problem.**
**あなたの問題を再現できる完全なソースコードを提示してください。**

```cpp
```
LGFX::LGFX() {
  auto cfg = bus.config();

  // SPI bus pins
  cfg.spi_host = VSPI_HOST;   // Use VSPI (GPIO 23, 18, 19)
  cfg.spi_mode = 0;
  cfg.freq_write = 27000000;
  cfg.freq_read  = 16000000;
  cfg.spi_3wire = false;      // SDA/MOSI only
  cfg.use_lock = true;

  cfg.pin_sclk = 18;   // SCL → GPIO18
  cfg.pin_mosi = 23;   // SDA → GPIO23
  cfg.pin_miso = -1;   // Not used
  cfg.pin_dc   = 4;    // D/CX → GPIO4

  bus.config(cfg);
  panel.setBus(&bus);

  auto pcfg = panel.config();
  pcfg.pin_cs   = 5;   // CSX → GPIO5
  pcfg.pin_rst  = 2;   // RESET → GPIO2
  pcfg.pin_busy = -1;  // Not used
  pcfg.invert = true;

  pcfg.panel_width  = 320;
  pcfg.panel_height = 480;
  panel.config(pcfg);

  setPanel(&panel);
}

LGFX tft;

void setup() {
  Serial.begin(115200);
  tft.init();
  delay(1000);
  tft.writecommand(0x10);
  delay(1000);
  tft.writecommand(0x11);
  delay(1000);
  tft.writecommand(0x28);
  delay(1000);
  tft.writecommand(0x29);
}

void loop() {
}
```

tobozo, 2025-08-12:
hi,

does it work better when enclosed between `startWrite()` and `endWrite()`?

```cpp
tft.startWrite();
tft.writecommand(0x10);
tft.endWrite();
```

tobozo, 2025-08-13:
startWrite/endWrite pull the CS pin high or low, this is needed otherwise the data is sent but ignored by the display