lovyan03/LovyanGFX #749: Device support request: ST7262 + GT911 display with CH422G I²C expander

igor-gia, 2025-08-20:
**Carefully written requests are more likely to be given priority.**  
**丁寧に記述された要望は優先して対応される可能性が高くなります。**

### Device Name (デバイスの名称・型番等)
7" RGB565 display with ST7262 controller, backlight via CH422G I/O expander

### URL of Device Specifications document (仕様書等のURL)
https://spotpear.com/wiki/ESP32-S3N8R8-7inch-LCD-Display-TouchScreen-800x480-LVGL-CAN-Sensor-RS485.html  

### URL of the store where we can purchase (商品を購入できるURL)
https://spotpear.com/shop/ESP32-S3N8R8-7inch-LCD-Display-TouchScreen-800x480-LVGL-CAN-Sensor-RS485/ESP32-S3-Touch-LCD-7.html  

---

### Problem (問題点)
This display board uses a **CH422G I/O expander** to control several signals, including the **backlight (DISP pin)**.  
Currently, I cannot find a ready-made configuration in LovyanGFX to handle this.  
Using the `ESP_IOExpander_Library` is not possible, because the touchscreen controller **GT911** also uses the same I²C bus, which leads to conflicts.  

### Request (要望内容)
Please add support in **LovyanGFX** for displays where the **backlight is controlled through a CH422G I/O expander**.  
Ideally, it should be possible to configure backlight control via CH422G without relying on external libraries.  

日本語で失礼いたします。  
このディスプレイでは CH422G IOエクスパンダを経由してバックライトを制御する必要がありますが、現状 LovyanGFX に適切な設定が見当たりません。  
GT911 タッチスクリーンと同じ I²C バスを共有しているため、既存の ESP_IOExpander ライブラリを使用すると競合が発生します。  
LovyanGFX 内で CH422G によるバックライト制御を設定できるよう、ご検討いただければ幸いです。  

### Environment (動作環境)
- MCU: **ESP32-S3**  
- Framework: **Arduino IDE**  
- Library: **LovyanGFX (latest release)**

igor-gia, 2025-08-22:
# LovyanGFX + CH422G Backlight Control

Thank you for the example code for controlling the backlight via the CH422G I²C expander alongside the GT911 touchscreen.

After several iterations of modifying the code, the following results were achieved:

* The screen now dims correctly when calling `setBrightness(0)` — the brightness decreases, although it does not turn off completely.
* If the backlight is manually turned OFF **before initializing LovyanGFX**, the screen is fully black and nothing is visible.

So, the proposed approach partially works, but full backlight control **after display initialization** is currently limited by shared I²C access with the touchscreen.

## 1️⃣ Minimal working example: `mail.ino`

```cpp
#include "Pins.h"

LGFX tft;

void setup()
{
    Serial.begin(115200);
    delay(100);

    Serial.println("Init display...");
    tft.init();
    tft.fillScreen(TFT_BLACK);
    tft.setTextColor(TFT_WHITE);
    tft.setTextSize(2);
    tft.setCursor(10, 10);
    tft.println("Hello LovyanGFX + CH422G backlight!");
}

void loop()
{
    // Demonstration of backlight ON/OFF
    tft.setBrightness(255); // ON
    delay(2000);
    tft.setBrightness(0);   // OFF
    delay(2000);
}
```

## 2️⃣ Full backlight driver: pins.h
```cpp
#define LGFX_USE_V1
#include <Wire.h>
#include <LovyanGFX.hpp>
#include <lgfx/v1/platforms/esp32s3/Panel_RGB.hpp>
#include <lgfx/v1/platforms/esp32s3/Bus_RGB.hpp>

#define TFT_HOR_RES   800
#define TFT_VER_RES   480

#define TOUCH_SDA 8
#define TOUCH_SCL 9
#define TOUCH_INT 4
#define TOUCH_RST -1

// CH422G expander pins (EXIO0..EXIO7)
#define TP_RST 1
#define LCD_BL 2
#define LCD_RST 3
#define SD_CS 4
#define USB_SEL 5

const uint8_t I2C_CH422G_ADDRESS = 0x24;

// ---------------------- Backlight via CH422G ----------------------
class IoExpandedBacklight : public lgfx::ILight {
public:
  void config(lgfx::ITouch::config_t *cfg) {
    if (cfg) {
      _cfg = *cfg;
      _wire = (_cfg.i2c_port == I2C_NUM_1) ? &Wire1 : &Wire;
      _i2c_addr = I2C_CH422G_ADDRESS;
    }
  }

  bool init(uint8_t brightness) override {
    setBrightness(brightness);
    return true;
  }

  void setBrightness(uint8_t brightness) override {
    if (!_wire) return;

    uint8_t data = (brightness > 0) ? 0xFF : 0x00;

    _wire->beginTransmission(_i2c_addr);
    _wire->write(data);
    _wire->endTransmission();
  }

private:
  lgfx::ITouch::config_t _cfg{};
  TwoWire* _wire = nullptr;
  uint8_t _i2c_addr = 0;
};
// -------------------------------------------------------------------

class LGFX : public lgfx::LGFX_Device {
public:
  lgfx::Bus_RGB      _bus_instance;
  lgfx::Panel_RGB    _panel_instance;
  IoExpandedBacklight _light_instance;
  lgfx::Touch_GT911  _touch_instance;

  LGFX(void) {
    {
      auto cfg = _panel_instance.config();
      cfg.memory_width  = TFT_HOR_RES;
      cfg.memory_height = TFT_VER_RES;
      cfg.panel_width   = TFT_HOR_RES;
      cfg.panel_height  = TFT_VER_RES;
      cfg.offset_x = 0;
      cfg.offset_y = 0;
      _panel_instance.config(cfg);
    }

    {
      auto cfg = _panel_instance.config_detail();
      cfg.use_psram = 1;
      _panel_instance.config_detail(cfg);
    }

    {
      auto cfg = _bus_instance.config();
      cfg.panel = &_panel_instance;

      cfg.pin_d0  = 14;
      cfg.pin_d1  = 38;
      cfg.pin_d2  = 18;
      cfg.pin_d3  = 17;
      cfg.pin_d4  = 10;

      cfg.pin_d5  = 39;
      cfg.pin_d6  = 0;
      cfg.pin_d7  = 45;
      cfg.pin_d8  = 48;
      cfg.pin_d9  = 47;
      cfg.pin_d10 = 21;

      cfg.pin_d11 = 1;
      cfg.pin_d12 = 2;
      cfg.pin_d13 = 42;
      cfg.pin_d14 = 41;
      cfg.pin_d15 = 40;

      cfg.pin_henable = 5;
      cfg.pin_vsync   = 3;
      cfg.pin_hsync   = 46;
      cfg.pin_pclk    = 7;
      cfg.freq_write  = 14000000;

      cfg.hsync_polarity     = 0;
      cfg.hsync_front_porch  = 20;
      cfg.hsync_pulse_width  = 10;
      cfg.hsync_back_porch   = 10;

      cfg.vsync_polarity     = 0;
      cfg.vsync_front_porch  = 10;
      cfg.vsync_pulse_width  = 10;
      cfg.vsync_back_porch   = 10;

      cfg.pclk_active_neg = 0;
      cfg.de_idle_high    = 0;
      cfg.pclk_idle_high  = 0;

      _bus_instance.config(cfg);
    }
    _panel_instance.setBus(&_bus_instance);

    {
      auto cfg = _touch_instance.config();
      cfg.x_min = 0;
      cfg.x_max = TFT_HOR_RES - 1;
      cfg.y_min = 0;
      cfg.y_max = TFT_VER_RES - 1;
      cfg.pin_int = TOUCH_INT;
      cfg.pin_rst = TOUCH_RST;
      cfg.bus_shared = false;
      cfg.offset_rotation = 0;
      cfg.i2c_port = I2C_NUM_1;
      cfg.pin_sda  = TOUCH_SDA;
      cfg.pin_scl  = TOUCH_SCL;
      cfg.freq     = 400000;
      cfg.i2c_addr = 0x14;
      _touch_instance.config(cfg);
      _panel_instance.setTouch(&_touch_instance);
    }

    {
      auto touch_cfg = _touch_instance.config();
      _light_instance.config(&touch_cfg);
      _panel_instance.setLight(&_light_instance);
    }

    setPanel(&_panel_instance);
  }
};
```
Maybe there are some other ideas?

tobozo, 2025-08-22:
using lgfx::i2c was the proposed solution to the i2c conflict, it doesn't cope well with Wire/Wire1

what happens when using this?

```cpp
  void setBrightness(uint8_t brightness) override {

    uint8_t data = (brightness > 0) ? 0xFF : 0x00;

    lgfx::i2c::transactionWrite(_cfg.i2c_port, _i2c_addr, &data, 1, _cfg.freq);

  }
```

tobozo, 2025-08-24:
interesting, so if the backlight pin is just an on/off switch, then `IoExpandedBacklight::init()` is complete and that backlight pin can be left alone, confirming that `IoExpandedBacklight::setBrightness()` should be using registers.

most CYD+LVGL demos seem to factor that situation to "enable or disable backlight only", but I'm quite sure it's possible to go further and control the brightness with more precision using the registers

According to the documentation 0x14 is a `W/R` command in the `command table 1`. I'm not sure what this means but it could be a clue.

[image]

Idea 1: 0x14 register wasn't poked properly

```cpp
  for (uint8_t i = 0; i <= 255; i += 8) {
    auto panel = (lgfx::Panel_RGB*)tft->getPanel();
    panel->writeCommand(0x14, 1);
    panel->writeData(i, 1);
  }
```

Idea 2: 0x4 register needs to be investigated along with regs 0x15 and 0x16

```cpp
  for (uint8_t i = 0; i <= 255; i += 8) {
    auto panel = (lgfx::Panel_RGB*)tft->getPanel();
    panel->writeCommand(0x14, 1);
    panel->writeData(i, 1);
    panel->writeData(0, 1); // value for reg 0x15
    panel->writeData(0, 1); // value for reg 0x16
  }
```

Idea 3: registers from command table 1 and 2 need to be sent through a specific I2C address?
Apparently there's 4 Command tables : `1`, `2`, `Gamma` and `Opts`, the latter has a default I2C value of 0x78, can you see such a device after an I2C scan? If si it could be worth trying to send the 0x14 register command to that device.

[image]

lightninlex, 2025-09-15:
Hi friends! Your adventure through this was incredible to read!
I am using the Waveshare esp32-s3 touch lcd 4.3b, and am having a lot of
problems. Could you please share your final code and configurations? I know
the pin definitions will probably be different, but it would be very helpful to see
How you are working with your board, as it uses the same components.