lovyan03/LovyanGFX #752: New RA8889 Display Development

gwiethaus, 2025-08-24:
I'm interested in contributing to the driver to support the RA8889 display. I recently purchased this display and it's very powerful, capable of displaying graphics and videos at high speed. It has SPI, I2C, and parallel communication. Its touchscreen is the new FT5316 capacitive microcontroller with 5 simultaneous touches.

I have basic code developed by Buydisplay and another by RAIO. I'd like to adapt these to the driver. I may need to create new functions in the base class that implements the RA8889 panel class.

I'd like to know where I can find the material described for code standardization and how I should proceed.

Can anyone give me some pointers on how I can use GitHub and post the new code as soon as a minimum working code for the display is ready?

AcuarioCat, 2025-08-25:
I did some work modifying a Teensy SPI driver from Teensy to ESP32 for this. 
https://github.com/wwatson4506/TeensyRA8889-SPI?tab=readme-ov-file
https://github.com/wwatson4506/TeensyRA8889-GFX-Common
I got it working with an ESP32 and a BuyDisplay RA8889 10" display but using SPI was way too slow for what I needed. 
I ran out of time to modify the RGB driver but it would be great if you do manage to get it working.

gwiethaus, 2025-08-25:
Hello @AcuarioCat 

I'm glad to hear that others were also developing for this processor. I'm just starting out. First, I'm studying how to adjust Visual Studio Code with the Github repository for checkout/checkin. This is my first time working with this method.

I'm also studying the code for how implementations are done in LovyanGFX. I think I'll need to study a little more advanced C++. I think I've managed to understand some things.

The difficulty will be moving from a driver model provided by Buydisplay to something more organized and structured like LovyanGFX.

I think I'll need some support to understand some things. I think it's complicated to always send messages and share code snippets, figures, and everything here. I see that this is more for clarifying questions than for more effective development interaction. I think it's bad to share something here. I always prefer email.

I'll check out your link. Your code. Actually, the new BuyDisplay ER-TFT070-2-6105 display is very fast and has a large internal memory of 128MB. I tested the touchscreen and some graphics. I found it very fast, even running on an Arduino UNO. Now imagine if it were on an ESP32-S3. I didn't have one because I didn't have an FFC to pin header adapter. I bought it and am waiting for it to arrive.

gwiethaus, 2025-08-25:
Hello again @AcuarioCat 

I checked part of your code; it was actually based on the RAIO code. There's also a newer version of the driver for the RA8889 on the BuyDisplay website that can also be used. Perhaps they've improved it. When I built the display, I tested it with the BuyDisplay code, and it worked flawlessly on an Arduino UNO using a BuyDisplay shield.

I believe that if you revisit the project, you'll have better success with ESP32, especially since both the RAIO website and BuyDisplay have provided more recent code for these displays, including ESP32.

gwiethaus, 2025-08-25:
Hello, Big @tobozo

Could you please create a step-by-step guide for me on how to integrate GitHub with Visual Studio Code? I think I should first download the repository using Git. I already installed Git and downloaded it using git clone.

I still haven't quite grasped how to check out and how to send and return what I want to write code on GitHub or take it there. And how can I do version control?

Could you instruct me on how to use this, how to install and configure it correctly in VSC?

I know I'm asking a lot, but I'm still a novice in this area. I've never used Git or version control. You can take your time; maybe this short guide will help others, and you can even write about this topic on one of your blogs.

If you need anything and I can help you and it is within my knowledge, it will be a pleasure.

closed 2025-10-01 by github-actions[bot]