lovyan03/LovyanGFX #754: fillGradientRect not working using HLINEAR

magicroomy, 2025-09-03:
I am working on a ESP32 (Wroom) using an LCD.
It is an Panel_ST7789 via SPI.
240x300 pixels
lovyan 1.2.7
Arduino (PlatformIO) Espressif 32 - 6.9.0

The Display Works. Text, Color Etc. All good.

Now I try to generate a Gradient Area (Horizontal).

I try to do it like this:

```cpp

static constexpr const lgfx::rgb888_t heatMap[]   = { {0, 0xff, 0}, {0xff, 0xff, 0}, {0xff, 0x80, 0}, {0xff, 0, 0}, {0x80, 0, 0} };
static constexpr const lgfx::colors_t heatMapGradient   = { heatMap,   sizeof(heatMap)/sizeof(lgfx::rgb888_t) };
tft.fillGradientRect(100, 100, 100, 100, heatMapGradient, lgfx::HLINEAR);
```

The result is that I get only a single line nevertheless with the gradient. It should be a rect.
If I change HLINEAR to VLINEAR I get a gradient rect 100x100... Vertical of course => as it should be

Do I do anything wrong here ?

magicroomy, 2025-09-03:
What can I say....
```cpp

......

LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 43 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Adafruit AS7341 @ 1.4.1
|-- **LovyanGFX @ 1.2.7**
Building in release mode
Compiling .pio\build\esp32dev\lib3af\LovyanGFX\lgfx\Fonts\IPA\lgfx_font_japan.c.o
Compiling .pio\build\esp32dev\lib3af\LovyanGFX\lgfx\Fonts\efont\lgfx_efont_cn.c.o
Compiling .pio\build\esp32dev\lib3af\LovyanGFX\lgfx\Fonts\efont\lgfx_efont_ja.c.o
Compiling .pio\build\esp32dev\lib3af\LovyanGFX\lgfx\Fonts\efont\lgfx_efont_kr.c.o
Compiling .pio\build\esp32dev\lib3af\LovyanGFX\lgfx\Fonts\efont\lgfx_efont_tw.c.o
Compiling .pio\build\esp32dev\lib3af\LovyanGFX\lgfx\utility\lgfx_miniz.c.o

...

```

tobozo, 2025-09-03:
oh right there was another change which isn't shipped with v1.2.7, it is on the develop branch though

```ini
lib_deps = 
    https://github.com/Loyvan03/LovyanGFX.git#develop
```

[image]

magicroomy, 2025-09-03:
I am afraid did did not work either.
Maybe it's a problem with the LCD ?
I tried a totally different one and it works perfectly as it should.
Hmmm

tobozo, 2025-09-03:
the clock speed seems a bit low (ST7789 can easily run at 40MHz, some variants at 80MHz) but that is probably unrelated

magicroomy, 2025-09-04:
I think it will be hard to pinpoint the reason for my problem so I close the issue right now. Especially since other LCDs work as expected.

closed 2025-09-04 by magicroomy