lovyan03/LovyanGFX #758: ESP32-P4 and ST7789 pushImage issue

cniedzi, 2025-09-13:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name:  Waveshare ESP32-P4-MODULE-DEV-KIT ([https://www.waveshare.com/esp32-p4-module-dev-kit.htm](https://www.waveshare.com/esp32-p4-module-dev-kit.htm))
- Panel Driver IC:    ST7789VW ([https://www.waveshare.com/product/displays/lcd-oled/lcd-oled-3/2inch-lcd-module.htm](https://www.waveshare.com/product/displays/lcd-oled/lcd-oled-3/2inch-lcd-module.htm)) 
- Bus type:          SPI
- LovyanGFX version:  latest (develop or master)
- FrameWork version:   ArduinoESP32 v3.3.0 | ESP-IDF v5.5.0
- Build Environment:   PlatformIO
- Operating System:   Windows

## Problem Description ( 問題の内容 )
When I try to display bitmaps smaller then 23 x 23 px using pushImage() they aren't displayed and the program freezes. When I comment respective pushImage command for small image, other bitmaps of the size at least 23 x 23 px are displayed, but they need to swap bytes for correct appearance (see the below image).
[image]
Unfortunately, the command setSwapBytes() won't work and causes the program to freeze.

The code works perfectly well for ESP32-S3 and the same ST7789 display (see the below image).
[image]

### Expected Behavior ( 期待される動作 )
1. All images are displayed correctly, no matter what size.
2. The function setSwapBytes(true) displays images correctly

### Actual Behavior ( 実際の動作 )
1. Images smaller then 23 x 23 px aren't displayed and the program freezes.
2. The function setSwapBytes(true) freezes the execution of the program.

### Steps to reproduce ( 再現のための前提条件 )
1. Connect ESP32-P4 to the ST7789 display
2. Compile and run the code

My wiring:
```
#define TFT_DC 22
#define TFT_RST 23
#define TFT_CS 26
#define TFT_BL 21
#define TFT_MOSI 32 
#define TFT_SCLK 36 
```

### Code to reproduce this issue ( 再現させるためのコード )

Config
```cpp

LGFX(void)
  {
    { 
      auto cfg = _bus_instance.config();
      cfg.spi_host = SPI2_HOST;
      cfg.spi_mode = 0;
      cfg.freq_write = 80000000;
      cfg.freq_read  = 16000000;
      cfg.spi_3wire  = false;
      cfg.use_lock   = true;
      cfg.dma_channel = SPI_DMA_CH_AUTO;
      cfg.pin_sclk = TFT_SCLK;
      cfg.pin_mosi = TFT_MOSI;
      cfg.pin_miso = TFT_MISO;
      cfg.pin_dc   = TFT_DC;
      _bus_instance.config(cfg);
      _panel_instance.setBus(&_bus_instance);
    }
    { 
        auto cfg = _panel_instance.config();
        cfg.pin_cs           = TFT_CS;
        cfg.pin_rst          = TFT_RST;
        cfg.panel_width      = 240;
        cfg.panel_height     = 320;
        cfg.invert           = true;
        cfg.rgb_order        = false;
        _panel_instance.config(cfg);
    }
    setPanel(&_panel_instance);
  }

```

Code
```cpp

// Dimensions    : 19x19 pixels
// Size          : 722 Bytes

const unsigned short Speaker[0x169] ={
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000,   // 0x0010 (16)
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x9492, 0xFFFF, 0x9492, 0x0000,   // 0x0020 (32)
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x94B2, 0xFFFF,   // 0x0030 (48)
0xFFFF, 0x8C71, 0x0000, 0x0000, 0x39C7, 0xD6BA, 0x4A49, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000,   // 0x0040 (64)
0x9CD3, 0xFFFF, 0xFFFF, 0xFFFF, 0x8C71, 0x0000, 0x0000, 0x2124, 0xEF7D, 0xFFFF, 0x2965, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000,   // 0x0050 (80)
0x0000, 0x0000, 0x9CF3, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0x8C71, 0x0000, 0x0000, 0x0000, 0x2945, 0xFFDF, 0xD6BA, 0x0020, 0x0000,   // 0x0060 (96)
0x4A69, 0xB596, 0xBDF7, 0xBDF7, 0xD69A, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0x8C51, 0x0000, 0x9CD3, 0x7BEF, 0x0000, 0x6B4D,   // 0x0070 (112)
0xFFFF, 0x632C, 0x6B4D, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0x8C51, 0x0000, 0xBDF7,   // 0x0080 (128)
0xFFFF, 0x4A49, 0x0000, 0xE73C, 0xDEDB, 0xF79E, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF,   // 0x0090 (144)
0x8C71, 0x0000, 0x0861, 0xEF7D, 0xDEDB, 0x0000, 0x8430, 0xFFFF, 0xF79E, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF,   // 0x00A0 (160)
0xFFFF, 0xFFFF, 0xFFFF, 0x8C71, 0x0000, 0x0000, 0x8430, 0xFFFF, 0x10A2, 0x39E7, 0xFFFF, 0xF79E, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF,   // 0x00B0 (176)
0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0x8C71, 0x0000, 0x0000, 0x528A, 0xFFFF, 0x39C7, 0x2104, 0xFFFF, 0xF79E, 0xFFFF,   // 0x00C0 (192)
0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0x8C71, 0x0000, 0x0000, 0x528A, 0xFFFF, 0x39C7, 0x2104,   // 0x00D0 (208)
0xFFFF, 0xF7BE, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0x8C71, 0x0000, 0x0000, 0x8C51,   // 0x00E0 (224)
0xFFFF, 0x10A2, 0x39E7, 0xFFFF, 0xF79E, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0x8C71,   // 0x00F0 (240)
0x0000, 0x0861, 0xEF7D, 0xD6BA, 0x0000, 0x8430, 0xFFFF, 0x5AEB, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF,   // 0x0100 (256)
0xFFFF, 0xFFFF, 0x8C51, 0x0000, 0xBDF7, 0xFFFF, 0x4228, 0x0000, 0xEF5D, 0xDEDB, 0x0000, 0x4208, 0xA534, 0xB596, 0xB596, 0xC638,   // 0x0110 (272)
0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0x8C51, 0x0000, 0x9492, 0x73AE, 0x0000, 0x6B4D, 0xFFFF, 0x630C, 0x0000, 0x0000, 0x0000,   // 0x0120 (288)
0x0000, 0x0000, 0x0000, 0x94B2, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0x8C71, 0x0000, 0x0000, 0x0000, 0x2945, 0xFFDF, 0xD6BA, 0x0020,   // 0x0130 (304)
0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x9492, 0xFFFF, 0xFFFF, 0xFFFF, 0x8C71, 0x0000, 0x0000, 0x2124, 0xF79E,   // 0x0140 (320)
0xFFFF, 0x2965, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x8C71, 0xFFFF, 0xFFFF, 0x8C71, 0x0000,   // 0x0150 (336)
0x0000, 0x31A6, 0xD6BA, 0x4228, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x8C51,   // 0x0160 (352)
0xFFFF, 0x9492, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000, };

tft.init();
tft.setRotation(3);
tft.setSwapBytes(true);

unsigned short *_speaker_psram = (unsigned short *) ps_malloc(sizeof(Speaker));
memcpy(_speaker_psram, Speaker, sizeof(Speaker));
if (_speaker_psram != NULL) pushImage(VOLUME_X_POS-25, VOLUME_Y_POS-19, 19, 19, _speaker_psram);

```

tobozo, 2025-09-13:
Hi,

Thanks for reporting this 👍 

Although the symptoms are different, it could be related to #743 hinting at an incomplete/improper handling of Bus_SPI with ESP32P4, which is being investigated.

Arduino core 3.3.0 introduced a lot of bugs with esp32-p4, and 3.2.1 is much more stable. this issue is probably consistent with both versions though.

tobozo, 2025-09-16:
hi,

tried with a 320*240 ILI9341 and an old 128*128 ST7789 but couldn't reproduce the issue. This narrows down the problem to the spi bus mode but doesn't help much :D 

Meanwhile I've ordered the Waveshare unit+display you linked earlier, hopefully I'll have for the weekend 🤞 

Can you setup a temporary project repository (or gist) with minimal code reproducing this issue so that we're on the same pinout/config basis?

tobozo, 2025-09-18:
unboxing complete \o/

I don't have a laser cutter so I had to improvise a hat 🤭 .. will now proceed to the wiring/testing

[image]

tobozo, 2025-09-18:
some thoughts on the current wiring:

- That display model does not have a MISO pin, which implies that `TFT_MISO` should be set to -1, not 33

- Pins 33/32 are pulled up (SH1.0 Connector), one of which is used in this configuration as MOSI

[image]

disabling miso and moving mosi to pin 20 didn't change anything though, now proceeding to investigate Bus_SPI

cniedzi, 2025-09-20:
Hi @tobozo, sorry for reaching out to you this way about a different issue, but maybe you might have some ideas:
[https://github.com/schreibfaul1/ESP32-audioI2S/issues/1167](https://github.com/schreibfaul1/ESP32-audioI2S/issues/1167).

tobozo, 2025-09-22:
@cniedzi triple-check your network and your code, this is a DNS problem, not a GFX or audio problem

cniedzi, 2025-09-22:
> [@cniedzi](https://github.com/cniedzi) triple-check your network and your code, this is a DNS problem, not a GFX or audio problem

@tobozo, already checked, the router reset to the firmware settings. No more network issues, but the problem with playing such streams still remains :( I can't understand, why everything is fine when I use Adafruit library... I have no clue where to start digging :)

cniedzi, 2025-09-22:
I'm curious about your results, you can use my project to check https://github.com/cniedzi/Test_AudioI2S_LovyanGFX/tree/main.

tobozo, 2025-09-22:
sorry I'm already busy with this issue and my own projects, also this is out of topic

ainyan03, 2026-08-25:
1.2.28 is out with the ESP32-P4 DMA fix from February plus a correction of the SPI clock divider base on P4 (the previous releases could run SCLK well below the requested rate). Could you re-test with 1.2.28 and let us know whether the issue is still there?

cniedzi, 2026-08-25:
Looks like it works.

```
     #ifdef ILI9488IPS
        cfg.spi_host = SPI2_HOST;     // ESP32-S2, C3, S3: SPI2_HOST or SPI3_HOST / ESP32 : VSPI_HOST or HSPI_HOST
        cfg.spi_mode = 0;
        cfg.freq_write = 79000000;
        cfg.freq_read  = 16000000;
        cfg.spi_3wire  = false;
        cfg.use_lock   = true;
        #ifdef ESP32P4
          cfg.dma_channel = SPI_DMA_CH_AUTO; //0;
        #else
          cfg.dma_channel = SPI_DMA_CH_AUTO;
        #endif
        cfg.pin_sclk = TFT_SCLK;
        cfg.pin_mosi = TFT_MOSI;
        cfg.pin_miso = TFT_MISO;
        cfg.pin_dc   = TFT_DC;
      #endif
```

[image]

[image]

closed 2026-08-28 by cniedzi